volkszaehler/mbmd #274: ORNO WE 516 wrong data to mqtt

mkmt, 2022-09-01:
Running command:

mbmd run -a /dev/tty.usb-1.1.4 -d ORNO3P:1 --raw --comset 8E1 -r 60s -m tcp://127.0.0.1:1883

gives me right values in web interface but completely wrong in mqtt, like:

...
homie/orno3p1-1/meter/voltagel1/$name L1 Voltage
homie/orno3p1-1/meter/voltagel1/$unit V
homie/orno3p1-1/meter/voltagel1/$datatype float
homie/orno3p1-1/meter/currentl1/$name L1 Current
homie/orno3p1-1/meter/currentl1/$unit A
...

Seems to be an error in mbmd code.

andig, 2022-09-02:
What is wrong? Observed value vs actual value?

mkmt, 2022-09-02:
@andig there is no value, instead of value I get what I think is name of variable, like in:

homie/orno3p1-1/meter/voltagel1/$name L1 Voltage

andig, 2022-09-02:
Value should be in 

    homie/orno3p1-1/meter/voltagel1

mkmt, 2022-09-02:
@andig yes it should but instead of valu in mqtt I get this:

...

mkmt, 2022-09-02:
Solved, just few first mqtt messages are like that, rest are normal.

By the way it is still a bug worth notiecing.

andig, 2023-01-31:
Closing as its not clear if that's an issue on mbmd side or related to the meter.

closed 2023-01-31 by andig