volkszaehler/mbmd #275: Kostal Plenticore 8.5 not found by scan

jarau-de, 2022-09-02:
Unfortunately the Plenticore 8.5 was not found by 'mbmd scan'.
192.168.0.30:502 -d kostal:71 is working. I found it in a forum.

andig, 2022-09-02:
Afaik KSEM is Sunspec. I don't think we're detecting Sunspec devices yet.

jarau-de, 2022-09-02:
But nevertheless it works great with mbmd. It would be nice to have the scan also working.

andig, 2026-09-30:
Sorry for letting this sit for so long without a resolution. Nobody has picked up the implementation in the meantime and there are no plans to work on it from the maintainer side, so I am closing this to keep the tracker honest. If the device is still relevant to you, a PR adding it (the existing meters/rs485/*.go files serve as templates) is very welcome and we will happily reopen.

🤖 Generated with [Claude Code](https://claude.com/claude-code)

closed 2026-09-30 by andig