volkszaehler/mbmd #276: mbmd ignores last digits after decimal

jarau-de, 2022-09-02:
mbmd ignores the last digits after decimal point for power, reactive power and apparent power. They are always zero. If I look on the webserver of smartmeter the digits are there.

andig, 2022-09-02:
Do you have an example? Which meter?

isarrider, 2022-09-02:
it is the same for me but I think it isn't related to mbmd, but to what gets published (in my case via Modbus TCP)...

jarau-de, 2022-09-02:
Yes indeed, it's rounded to 10 W. It would be nice to have the full resolution.

andig, 2022-09-02:
> but I think it isn't related to mbmd

Closing here- we can only output what the device offers...

isarrider, 2022-09-07:
@jarau-de can you be sure that the inverter provides these granular data via Modbus TCP?
(Dont get me wrong, I would like to have it too - 10W jumps suck ;))

jarau-de, 2022-09-07:
Ok, obviously there was a confusion on my side between KSEM and Kostal Plenticore. Both of them deliver Informations using Modbus TCP, but over different ports (502, 1502) and of course different IP.
It is true, as mentioned by andig, the Sunspec values by KSEM of power are given by a resolution of 10^1 W. It is well defined in the KSEM documentation.
[image]

 But it has a lot more registers. See table below.

[image]

These registers have a much better resolution of 0.1 W. I have tested that with pyModbusTCP. But I don't know whether it could or would be implemented in mbmd... For my personal use I read the electric meter of my house directly by optical interface and volkszaehler. That means I don't need KSEM for this purpose.

But much more interesting would be the Modbus TCP interface of the Plenticore inverter. It delivers the total DC and AC power as well as the home consumption of inverter power which I need for tax office (and many, many other parameters). It would be very interesting to see these values in mbmd. I tried that using
./mbmd run -a 192.168.0.31:1502 -d KOSTAL:71
but unfortunately the values look a bit strange. I think it is caused by a different byte order?. It uses little endian (default) instead of big endian (SunSpec). I don't want to touch that configuration. For pyModbusTCP I had to switch the order to get proper values.

[image]
If mbmd would implement these values I would be happy.

andig, 2022-09-07:
> It is true, as mentioned by andig, the Sunspec values by KSEM of power are given by a resolution of 10^1 W. It is well defined in the KSEM documentation.

I can't remember saying that. The Sunspec models have scale factors for basically every model/block/point which provide arbitrary resolution.

I'm lost: what device and register are we talking about now? Sunspec or not?

jarau-de, 2022-09-07:
@andig 
"Closing here- we can only output what the device offers..."
This what you said and you are right. :-)

I would like to see my Kostal Plenticore power SunSpec informations correctly (byte order).

Other users want to have higher resolution of KSEM power informations, but higher resolution seems not suppored by SunSpec registers. If mbmd concentrates on SunSpec, then KSEM question was answered, I think.

jarau-de, 2022-09-08:
Good news, I switched the Plenticore to SunSpec byte order. Now mbmd works as expected and I can put inverter power into vz db. Now I have all I need and for me this topic is closed.

closed 2022-09-08 by jarau-de

isarrider, 2022-09-08:
Hi - can you detail a bit more what you did?