volkszaehler/mbmd #285: SolarEdge Inverter: sunspec: subdevice 1 not found

cnovak, 2022-10-07:
When starting evcc, I get the following error:

`[main  ] FATAL 2022/10/06 21:54:49 cannot create meter 'grid1': cannot create meter 'template': cannot create meter 'custom': power: sunspec: subdevice 1 not found`

I configured my system to use my SolarEdge inverter. I can see it, and the `evcc detect` brings it up:

```
+---------------+----------+---------+---------------------------------------------------------------------------------------------------------+
|      IP       | HOSTNAME |  TASK   |                                                 DETAILS                                                 |
+---------------+----------+---------+---------------------------------------------------------------------------------------------------------+
| 192.168.86.84 |          | sunspec | {"IP":"192.168.86.84","Port":1502,"ModbusResult":{"SlaveID":1,"Model":1,"Point":"Mn","Value":"SolarEdge |
|               |          |         | "}}                                                                                                     |
+---------------+----------+---------+---------------------------------------------------------------------------------------------------------+
```

The configuration I have for it in evcc.yaml is: 

```
meters:
- type: template
  template: solaredge-inverter 
  id: 1
  host: 192.168.86.84  
  port: 1502  
  usage: grid  
  modbus: tcpip  
  name: grid1
- type: template
  template: solaredge-inverter 
  id: 102  
  host: 192.168.86.84  
  port: 1502  
  usage: pv  
  modbus: tcpip  
  name: pv2

```

It looks like the issue is coming from the `https://github.com/volkszaehler/mbmd`, file: [sunspec.go#L76](https://github.com/volkszaehler/mbmd/blob/f715cedbb7892ed51f80f067465bdd8261cf7304/meters/sunspec/sunspec.go#L76)

`d.subdevice` is: `1`
`devices` has 1 item in the array

Is there a missing config attribute I need to add?

premultiply, 2022-10-07:
Have you connected a grid meter to your inverter at all?

cnovak, 2022-10-07:
*NOTE* Made a copy of this issue at EVCC project: https://github.com/evcc-io/evcc/issues/4749, maybe this is more applicable there versus here?

closed 2022-10-07 by andig