volkszaehler/mbmd #286: Arch Linux package (AUR) created

archi, 2022-10-07:
I wanted to let you know that I created and pushed a package to the Arch Linux AUR: https://aur.archlinux.org/packages/mbmd-git

I did not like the example service file (sorry ;-)), and adjusted it. The security is a bit tighter now, and the daemon runs as its own user. Sadly Amazon only delivers the USB RS485 adapter tomorrow, so I can not really test it then.

If there is any feedback, I'm happy to incorporate it. Also, in case you'd like to link to it in the README, I'll be happy to open a small PR.

btw, thanks for all the work! Looking forward to feed my new SDM630 into Home Assistant!

archi, 2022-10-10:
FTR: Reading my SDM630 works as expected :)

isarrider, 2022-10-21:
> FTR: Reading my SDM630 works as expected :)

could you post your config?
A friend has this meter too...

archi, 2022-10-21:
@isarrider for mbmd it's really just https://github.com/volkszaehler/mbmd/blob/master/mbmd.dist.yaml minus the things I don't need. For Home Assistant I've used an mqtt client and manually added all the fields I felt relevant.

I've created a gist with my configs https://gist.github.com/archi/320d21923e26aedf49ebc2e34659f7da

isarrider, 2022-10-24:
@archi thx!

andig, 2026-09-30:
Thanks for packaging mbmd for Arch. The AUR package is a downstream matter and there is nothing to change in this repository, so closing this. A PR adding a link to the README is still welcome.

🤖 Generated with [Claude Code](https://claude.com/claude-code)

closed 2026-09-30 by andig

archi, 2026-09-30:
I'd be happy to also create a PR for the more secure service file, if you're interested.

Also nice you're checking your backlog :)

andig, 2026-09-30:
Yeah, had lazy Claude do the most urgent chores…