volkszaehler/mbmd #289: runtime error bei Gerät IEM3000

Muckiy, 2022-10-23:
Hallo,

ich habe ebenfalls ein Problem mit dem iem3000 Gerät (bei mir iem3155).

`./mbmd -v --raw run --devices iem3000:1 -a /dev/ttyUSB0
2022/10/23 17:05:33 mbmd 0.13 (fa6b574)
2022/10/23 17:05:33 config: creating RTU connection for /dev/ttyUSB0 (9600baud, 8N1)
2022/10/23 17:05:33 config: declared device IEM3000:1.0
2022/10/23 17:05:33 httpd: starting api at 0.0.0.0:8080
2022/10/23 17:05:33 initialized device IEM30001.1: {IEM3000 IEM3000 Schneider Electric iEM3000 series    0}
2022/10/23 17:05:33 modbus: send 01 03 0d c2 00 02 67 5b
2022/10/23 17:05:33 modbus: recv 01 03 04 00 00 00 00 fa 33
panic: runtime error: index out of range [7] with length 4

goroutine 52 [running]:
encoding/binary.bigEndian.Uint64(...)
        /home/travis/.gimme/versions/go1.13.linux.amd64/src/encoding/binary/binary.go:124
github.com/volkszaehler/mbmd/meters/rs485.RTUInt64ToFloat64(0x2d53303, 0x4, 0xfd, 0x2ddc3c0, 0x2c25300)
        /home/travis/gopath/src/github.com/volkszaehler/mbmd/meters/rs485/transform.go:76 +0x88
github.com/volkszaehler/mbmd/meters/rs485.MakeScaledTransform.func1(0x2d53303, 0x4, 0xfd, 0x4, 0xfd)
        /home/travis/gopath/src/github.com/volkszaehler/mbmd/meters/rs485/transform.go:83 +0x3c
github.com/volkszaehler/mbmd/meters/rs485.(*RS485).QueryOp(0x2d99b20, 0x8e1360, 0x2c6f0d0, 0xdc20003, 0x2, 0x30, 0x2ec4100, 0x0, 0x0, 0x0, ...)
        /home/travis/gopath/src/github.com/volkszaehler/mbmd/meters/rs485/rs485.go:101 +0xd4
github.com/volkszaehler/mbmd/meters/rs485.(*RS485).Query(0x2d99b20, 0x8e1360, 0x2c6f0d0, 0xa, 0x2ec2048, 0xa, 0xb, 0x66daafff)
        /home/travis/gopath/src/github.com/volkszaehler/mbmd/meters/rs485/rs485.go:137 +0xe4
github.com/volkszaehler/mbmd/server.(*Handler).queryDevice(0x2c6f130, 0x8dd508, 0x2d99d80, 0x2c5f1c0, 0x2c5f180, 0x8dd901, 0x8dd948, 0x2d99b20)
        /home/travis/gopath/src/github.com/volkszaehler/mbmd/server/handler.go:136 +0xd8
github.com/volkszaehler/mbmd/server.(*Handler).Run.func1(0x2c5c001, 0x8dd948, 0x2d99b20)
        /home/travis/gopath/src/github.com/volkszaehler/mbmd/server/handler.go:84 +0x14c
github.com/volkszaehler/mbmd/meters.(*Manager).All(0x2d99a80, 0x2e78f40)
        /home/travis/gopath/src/github.com/volkszaehler/mbmd/meters/manager.go:42 +0x6c
github.com/volkszaehler/mbmd/server.(*Handler).Run(0x2c6f130, 0x8dd508, 0x2d99d80, 0x2c5f1c0, 0x2c5f180)
        /home/travis/gopath/src/github.com/volkszaehler/mbmd/server/handler.go:55 +0x60
github.com/volkszaehler/mbmd/server.(*QueryEngine).Run.func1(0x3b9aca00, 0x0, 0x8dd508, 0x2d99d80, 0x2c5f1c0, 0x2c5f180, 0x2d3e010, 0x2c6f130)
        /home/travis/gopath/src/github.com/volkszaehler/mbmd/server/queryengine.go:95 +0x80
created by github.com/volkszaehler/mbmd/server.(*QueryEngine).Run
        /home/travis/gopath/src/github.com/volkszaehler/mbmd/server/queryengine.go:89 +0x128
`

Falls du einen Lösungsansatz weißt oder noch etwas brauchst, melde dich bitte.

Danke im Vorraus.

Muckiy

andig, 2022-10-23:
Da scheint ein Register falsch definiert zu sein- 32bit gelesen, aber 64bit erwartet.

andig, 2022-10-23:
Wenn ich https://ckm-content.se.com/ckmContent/sfc/servlet.shepherd/document/download/0691H00000Dwo0bQAB mit der aktuellen Implementierung vergleiche dann scheint es, als wären alle Registeradressen um 1 zu niedrig. Bin mir nicht sicher, ob überhaupt schonmal jemand mit diesem Zähler getestet hat?