volkszaehler/mbmd #291: cannot allocate memory after two days

chrissooo, 2022-10-26:
Ich verwende mbmd folgendermaßen:
/usr/bin/mbmd read -d ORNO1p:1 -a /dev/ttyUSB2 -b 9600 --comset 8E1 40961 1

Es funktioniert auch eigentlich ganz gut. Der Wert kommt ganz normal raus. Mir ist das Problem, dann wenn diese Anfrage ca. 2 Tage läuft (wird jede 7 Sekunden ausgeführt), dann kommt nur noch:
cannot allocate memory

Wenn dir den PI neu starte, funktioniert es wieder für ca. 2 Tage.

Kann mir jemand helfen?

chrissooo, 2022-10-26:
Nachtrag,

Der USB Adapter scheint das Problem zu sein:
pi@nas:~ $ cat /dev/ttyUSB1
pi@nas:~ $ cat /dev/ttyUSB0
pi@nas:~ $ cat /dev/ttyUSB2
cat: /dev/ttyUSB2: Nicht genügend Hauptspeicher verfügbar

USB2 ist Tod. Nach Neustart geht es wieder für zwei Tage. Tut mbmd hier irgendwas beim abfragen nicht beenden oder so?

chrissooo, 2022-10-26:
Hier noch die Aufgabe zum Speicher, sieht gut aus:
pi@nas:~ $ free -h
              gesamt       benutzt     frei      gemns.  Puffer/Cache verfügbar
Speicher:

chrissooo, 2022-10-26:
Noch ein Nachtrag, das ganze kann ich loswerden wenn ich den Adapter neu initialisiere. Das werde ich jetzt ertsmal als Cronjob alle 24 Stunden machen. Wie gesagt, passiert das ja rund alle 2 Tage.

andig, 2022-10-26:
Offensichtlich ein Memoryleak. Die lassen sich mit Go ganz gut diagnostizieren. Wenn Du das Interesse hast dabei zu helfen baue ich die Unterstützung für die Diagnose mit ein, dann kannst Du mit tracen und wir die Traces anschließend auswerten.

chrissooo, 2022-10-27:
Gern kann ich das machen. Bauen kann ich selber wenn du etwas vorbereitest.

andig, 2022-10-27:
Probiers mal. Siehe https://github.com/evcc-io/evcc/issues/2748#issuecomment-1055626609. Hier sollte `memprofile` zur Anwendung kommen.

chrissooo, 2022-10-28:
Hi, ich werde das erst nächste Woche testen könne. Aber danke schonmal. Will aber noch was berichten, bzw. ergängen.

Das ganze rufe ich über exec() in einem PHP Skript auf. Das ganze läuft auf einem lighttpd Server. Der Aufruf wird alle 7 Sekunden von InfluxDB gemacht. Soweit die Umstände, die habe ja noch gar nicht so geschildert.

Jetzt habe ich vorgestern ein anderes Tool verwendet statt mbmd. Konkret mbpoll womit ich auch Modus Adressen abfragen kann. Das ganze auch in dem PHP Skript über exec(). Jetzt war heute der USBStick auch wieder auf dem gleichem Zustand. 

Das Neustarten von lighttpd oder php hat nichts gebracht. Auch free zeigt nichts auffälliges, noch jede Menge freier Speicher. 

Ich werde deine Version auf jeden Fall laufen lassen um zu sehen was passiert, aber vielleicht liegt es am Betriebssystem? Das ist halt ein Rasparry PI 4 mit Standard OS... ich berichte.

chrissooo, 2022-10-28:
Jetzt bin ich doch remote drauf und habe gebaut. Jetzt check ich es gerade nicht wie ich das debugen soll. Ich rufe das ja immer so auf:
`mbmd read -d ORNO1p:1 -a /dev/ttyUSB2 -b 9600 --comset 8E1 40961 1`
Dann zeigt er mir das Ergebnis und das Programm wird beendet.

--profile oder --memprofile anhängen sagt dass es den parameter nicht gibt ...

andig, 2022-10-28:
Das macht in Deinem Fall (der erst jetzt klar wird) auch keinen Sinn. `read` kann kein Speicherleck haben da der Prozess ja immer wieder beendet wird. Dein Problem hat also nix mit mbmd zu tun.

chrissooo, 2022-10-28:
Ok, verstehe ... das sorry für die Umstände. Der Workaround mit dem nächlichen reinitialisieren des Sticks geht ja .. Danke!

andig, 2022-10-28:
Macht nix, das Feature können wir ja trotzdem mal brauchen...

closed 2022-10-28 by andig