volkszaehler/mbmd #296: Insertion request Aurora PVI-2000-OUTD-IT

stepsolar, 2022-11-19:
Hi, congratulations on your project. I ask you if it would be possible to insert the "Aurora PVI-2000-OUTD-IT" inverter. I have made several attempts with the abb parameter but it is not detected. Thank you

andig, 2022-11-19:
Sure, PR welcome

stepsolar, 2022-11-19:
I found this file https://www.drhack.it/images/PDF/AuroraCommunicationProtocol_4_2.pdf

letsrock303, 2023-01-09:
Hello,

I'm interested as well in an integration of ABB/Powerone PVI AUrora Inverter.
I found information about how it can work:
http://www.curtronics.com/Solar/AuroraData.html
This tool works fine, but it would be better to have an integration in mbmd.

thanks
KR

andig, 2026-09-30:
The Aurora inverters speak the proprietary Power-One serial protocol rather than Modbus, which is outside the scope of mbmd, and no PR has materialised since 2022. Closing; happy to reopen if someone wants to contribute a Modbus-based implementation.

🤖 Generated with [Claude Code](https://claude.com/claude-code)

closed 2026-09-30 by andig