volkszaehler/mbmd #297: EM24 issues with Opcodes and negative power values

korpa, 2022-12-09:
I tried to use mbmd to read data from an RS485 EM24. Sadly a few values where totally wrong. After digging into the documentation I was able to fix the Opcodes like this:

```diff
@@ -27,16 +27,16 @@ func NewCarloGavazziProducer() Producer {
                PowerL2:   0x14,
                PowerL3:   0x16,
                Power:     0x28,
-               CosphiL1:  0x2E,
-               CosphiL2:  0x2F,
-               CosphiL3:  0x30,
-               Cosphi:    0x31,
-               Frequency: 0x33,
-               Import:    0x34,
-               ImportL1:  0x40,
-               ImportL2:  0x42,
-               ImportL3:  0x44,
-               Export:    0x4E,
+               CosphiL1:  0x32,
+               CosphiL2:  0x33,
+               CosphiL3:  0x34,
+               Cosphi:    0x35,
+               Frequency: 0x37,
+               Import:    0x42,
+               ImportL1:  0x46,
+               ImportL2:  0x48,
+               ImportL3:  0x4A,
+               Export:    0x5C,
```

Now most values written to MQTT seem to be ok. 

But power values never get negative, even if power flow is reversed through high solar power in the system. I could not verify if this issue is a setting issue within the EM24 or if the number encoding of mbmd (`Int32LswFirst`) never returns negative values.

andig, 2022-12-09:
Looking at https://www.aggsoft.com/serial-data-logger/tutorials/modbus-data-logging/carlo-gavazzi-em24.htm it seems that the registers are currently wrong. Funny enough, they seemed to be right before https://github.com/volkszaehler/mbmd/pull/260. I can no longer reproduce why that change was made. I think we could just go back to the original register map which was copied 1:1 from the docs.

Would that work for you? Do you want to open a PR?

korpa, 2022-12-09:
Why do you look at this aggsoft URL and not into the official documentation. I provides much more information than the aggsoft URL (e.g. the factors):

[image]

Regarding the Opcodes, I can provide the fix via PR. 

But the issue regarding encoding of the power values. My knowledge about bit operations is only very basic. Therefore my question: Do you think switching back from `RTUInt32ToFloat64Swapped` to `RTUInt32ToFloat64` will solve the issue, that power never gets negative?

[image]

korpa, 2022-12-09:
By the way. The original documentation could be found here: https://gavazzi.se/app/uploads/2020/11/em24_is_cp.pdf

andig, 2022-12-09:
> Why do you look at this aggsoft URL and not into the official documentation.

Probably because I didn't find it at time of writing.

> Do you think switching back from RTUInt32ToFloat64Swapped to RTUInt32ToFloat64 will solve the issue, that power never gets negative?

I don't think so but you could just give it a try. If wrong you'll notice immediately looking at the 2-word decodings.

korpa, 2022-12-09:
I tested `RTUInt32ToFloat64`. This returns totally wrong readings. So we need the swapped version.

As I said, I'm not very familiar with bit operations. I have no idea how to change `Int32LswFirst` that it supports negative values as well.

andig, 2022-12-09:
> As I said, I'm not very familiar with bit operations. I have no idea how to change Int32LswFirst that it supports negative values as well.

The meter is probably not sending them. Enable logging to see wire format.

korpa, 2022-12-23:
Fixed in  #298

closed 2022-12-23 by korpa