volkszaehler/mbmd #308: Huawei SUN2000 Inverter

nicx, 2023-03-15:
Hi, just amshort question: Is it possible to use MBMD wit a Huawei SUN2000 inverter in any way? Imdont found a hint in the documentation...

nicx, 2023-03-20:
any hint? :)

andig, 2023-03-20:
If it supports Sunspec yes, otherwise out of scope

nicx, 2023-03-20:
@andig thanks for your response. I thought EVCC is using this project as "underlaying connection lib", and because with EVCC the connection to my non-sunspec-inverter Huawei works, I thought MBMD should work directly, too. Anyway, thanks again.

wexman, 2024-05-30:
I came here for the same question, but it's not really answered, is it?

I also thought EVCC was using the mbmd source for accessing the huawei inverters, so that should be working. But I wasn't able to get mbmd itself to work.

Can we clearify this? Does it work or does it not?