volkszaehler/mbmd #309: Polling interval of the Meter customizable

iz8mbw, 2023-03-23:
Hello.
I'm using mbmd since few days so I'm not expert on this.
My goal is to use mbmd with Home Assistant and may be in future publish the data of my SDM120 meter to MQTT.

I see for now mbmd publish data via a Web interface and for now it's OK for me since I can interrogate the Web interface via various way (curl is an example),

My question for now is: is it possible to customize via a parameter the Energy Meter (SDM120 in my case) polling interval?

Thanks.

grmat, 2023-07-18:
To poll every 30 seconds for example, use the cli parameter "rate " (e.g. `--rate 30s`) or add `rate: 30s`" at the top level in your yaml config.

andig, 2026-09-30:
mbmd has a global poll rate limit: pass `--rate 30s` on the command line or add `rate: 30s` at the top level of mbmd.yaml. Closing as answered; please reopen if that does not cover your case.

🤖 Generated with [Claude Code](https://claude.com/claude-code)

closed 2026-09-30 by andig