volkszaehler/mbmd #311: TCP Connection ends in timeout - Serial connection works fine

ChristianSteffens, 2023-03-25:
I'm using your library for quite some time with Eastron meters (3 x SDM 630, 2 x SDM 230) over a regular wired RS485 connection. This works very well, even for a long period of time, without any interruptions.

Now I'd like to migrate this setup to an Ethernet / TCP based connection. For this I bought a [Modbus TCP Gateway from Waveshare](https://www.waveshare.com/wiki/RS485_TO_POE_ETH_(B)#Documentation). This device allows transparent serial <> ethernet connections **or** converts Modbus RTU to Modbus TCP.

To keep things simple I tried using the transparent serial <> ethernet connection first, i.e. just use a MyIp:MyPort device in MBMD and setting the property `rtu = true`. Unfortunately no data is then aquired. I only get some timeout and here and then a CRC error:

[image]

```
mbmd_1  | 2023/03/25 23:04:28 mbmd 0.13 (fa6b574)
mbmd_1  | 2023/03/25 23:04:28 config: using /etc/mbmd.yaml
mbmd_1  | 2023/03/25 23:04:28 config: creating RTU over TCP connection for 192.168.1.7:502
mbmd_1  | 2023/03/25 23:04:28 config: using default adapter 192.168.1.7:502 for device {sdm 1 0 home }
mbmd_1  | 2023/03/25 23:04:28 config: using default adapter 192.168.1.7:502 for device {sdm 2 0 office }
mbmd_1  | 2023/03/25 23:04:28 config: using default adapter 192.168.1.7:502 for device {sdm 3 0 EV }
mbmd_1  | 2023/03/25 23:04:28 config: using default adapter 192.168.1.7:502 for device {sdm230 4 0 WR2 }
mbmd_1  | 2023/03/25 23:04:28 config: using default adapter 192.168.1.7:502 for device {sdm230 5 0 WR1 }
mbmd_1  | 2023/03/25 23:04:28 config: declared device SDM:1.0
mbmd_1  | 2023/03/25 23:04:28 config: declared device SDM:2.0
mbmd_1  | 2023/03/25 23:04:28 config: declared device SDM:3.0
mbmd_1  | 2023/03/25 23:04:28 config: declared device SDM230:4.0
mbmd_1  | 2023/03/25 23:04:28 config: declared device SDM230:5.0
mbmd_1  | 2023/03/25 23:04:28 httpd: starting api at 0.0.0.0:8080
mbmd_1  | 2023/03/25 23:04:28 mqtt: connecting mbmd at [tcp://192.168.1.4:1883]
mbmd_1  | 2023/03/25 23:04:28 initialized device SDM1.1: {SDM SDM Eastron SDM630    0}
mbmd_1  | 2023/03/25 23:04:28 modbus: send 01 04 00 0c 00 02 b1 c8
mbmd_1  | 2023/03/25 23:04:29 device SDM1.1 did not respond (1/3): read failed: read tcp 172.19.0.2:59072->192.168.1.7:502: i/o timeout
mbmd_1  | 2023/03/25 23:04:30 modbus: send 01 04 00 0c 00 02 b1 c8
mbmd_1  | 2023/03/25 23:04:31 device SDM1.1 did not respond (2/3): read failed: read tcp 172.19.0.2:59072->192.168.1.7:502: i/o timeout
mbmd_1  | 2023/03/25 23:04:32 modbus: send 01 04 00 0c 00 02 b1 c8
mbmd_1  | 2023/03/25 23:04:33 device SDM1.1 did not respond (3/3): read failed: read tcp 172.19.0.2:59072->192.168.1.7:502: i/o timeout
mbmd_1  | 2023/03/25 23:04:34 device SDM1.1 is offline
mbmd_1  | 2023/03/25 23:04:34 initialized device SDM1.2: {SDM SDM Eastron SDM630    0}
mbmd_1  | 2023/03/25 23:04:34 modbus: send 02 04 00 3c 00 02 b1 f4
mbmd_1  | 2023/03/25 23:04:34 modbus: recv 02 10 c3 8a 4d 13
mbmd_1  | 2023/03/25 23:04:34 device SDM1.2 did not respond (1/3): read failed: modbus: response crc '4941' does not match expected '3793'
mbmd_1  | 2023/03/25 23:04:35 modbus: send 02 04 00 3c 00 02 b1 f4
mbmd_1  | 2023/03/25 23:04:35 modbus: recv 02 10 c3 aa f7 52 5b
mbmd_1  | 2023/03/25 23:04:35 device SDM1.2 did not respond (2/3): read failed: modbus: response crc '23378' does not match expected '6806'
mbmd_1  | 2023/03/25 23:04:36 modbus: send 02 04 00 3c 00 02 b1 f4
mbmd_1  | 2023/03/25 23:04:36 modbus: recv 02 10 c3 53 89 78 fd
mbmd_1  | 2023/03/25 23:04:36 device SDM1.2 did not respond (3/3): read failed: modbus: response crc '64888' does not match expected '27220'
```

Changing the `rtu = false` parameter and enable the translation `Modbus TCP` in the Waveshare Gateway gives only timeouts:

[image]

```
mbmd_1  | 2023/03/25 23:09:54 mbmd 0.13 (fa6b574)
mbmd_1  | 2023/03/25 23:09:54 config: using /etc/mbmd.yaml
mbmd_1  | 2023/03/25 23:09:54 config: creating TCP connection for 192.168.1.7:502
mbmd_1  | 2023/03/25 23:09:54 config: using default adapter 192.168.1.7:502 for device {sdm 1 0 home }
mbmd_1  | 2023/03/25 23:09:54 config: using default adapter 192.168.1.7:502 for device {sdm 2 0 office }
mbmd_1  | 2023/03/25 23:09:54 config: using default adapter 192.168.1.7:502 for device {sdm 3 0 EV }
mbmd_1  | 2023/03/25 23:09:54 config: using default adapter 192.168.1.7:502 for device {sdm230 4 0 WR2 }
mbmd_1  | 2023/03/25 23:09:54 config: using default adapter 192.168.1.7:502 for device {sdm230 5 0 WR1 }
mbmd_1  | 2023/03/25 23:09:54 config: declared device SDM:1.0
mbmd_1  | 2023/03/25 23:09:54 config: declared device SDM:2.0
mbmd_1  | 2023/03/25 23:09:54 config: declared device SDM:3.0
mbmd_1  | 2023/03/25 23:09:54 config: declared device SDM230:4.0
mbmd_1  | 2023/03/25 23:09:54 config: declared device SDM230:5.0
mbmd_1  | 2023/03/25 23:09:54 httpd: starting api at 0.0.0.0:8080
mbmd_1  | 2023/03/25 23:09:54 mqtt: connecting mbmd at [tcp://192.168.1.4:1883]
mbmd_1  | 2023/03/25 23:09:54 initialized device SDM1.1: {SDM SDM Eastron SDM630    0}
mbmd_1  | 2023/03/25 23:09:54 modbus: send 00 01 00 00 00 06 01 04 00 04 00 02
mbmd_1  | 2023/03/25 23:09:55 device SDM1.1 did not respond (1/3): read failed: read tcp 172.19.0.2:35306->192.168.1.7:502: i/o timeout
mbmd_1  | 2023/03/25 23:09:56 modbus: send 00 02 00 00 00 06 01 04 00 04 00 02
mbmd_1  | 2023/03/25 23:09:57 device SDM1.1 did not respond (2/3): read failed: read tcp 172.19.0.2:35306->192.168.1.7:502: i/o timeout
mbmd_1  | 2023/03/25 23:09:58 modbus: send 00 03 00 00 00 06 01 04 00 04 00 02
mbmd_1  | 2023/03/25 23:09:59 device SDM1.1 did not respond (3/3): read failed: read tcp 172.19.0.2:35306->192.168.1.7:502: i/o timeout
mbmd_1  | 2023/03/25 23:10:00 device SDM1.1 is offline
```

Since the physical setup works fine (changing the connection back to the serial RS485 USB adapter works perfectly), I wouldn't assume a hardware issue, but more likely a configuration or software problem in general.

Any ideas on how to investigate furthere here?

ChristianSteffens, 2023-04-01:
Increased the timeout - and now it works - maybe this should be mentioned as a common fix in the README?

closed 2023-04-01 by ChristianSteffens

holli73, 2024-10-01:
> Increased the timeout - and now it works - maybe this should be mentioned as a common fix in the README?

hello,

i have the same problem - but i can't get the timeout parameter working - where did you place it?
it is not working on the command line and not within my yaml as well - i tried to place it in every section but still everytime i get a message that this parameter is not known:

f.ex.:

```
./mbmd --timeout 500ms run -c $PWD/mbmd.yaml
unknown flag: --timeout

```
or within the yaml

```
# REST api, use 127.0.0.1 to restrict to localhost
api: 0.0.0.0:8081
#timeout: 700ms

# adapters are referenced by device
adapters:
- device: 192.168.245.44:502
  rtu: true # Modbus RS485 to Ethernet converter uses RTU over TCP
#  timeout: 700ms

# list of devices
devices:
- name: sdm
  type: sdm
  id: 1
  adapter: 192.168.245.44:502
  timeout: 700ms
```
results in
```
holli@rasp2-verteiler-kasten:~/BUILD/ModBusRs485/mbmd $ ./mbmd run -c $PWD/mbmd.yaml
2024/10/01 20:02:57 mbmd 0.13 (fa6b574)
2024/10/01 20:02:57 config: using /home/holli/BUILD/ModBusRs485/mbmd/mbmd.yaml
2024/10/01 20:02:57 config: failed parsing config file /home/holli/BUILD/ModBusRs485/mbmd/mbmd.yaml: 1 error(s) decoding:

* 'Devices[0]' has invalid keys: timeout
```
any hint would be great 
thanks
holli