volkszaehler/mbmd #317: Fix Support for CG EM24 Ethernet Version

hrpv, 2023-05-09:
The EM24 Ethernet Version has a different register mapping than the EM24 RS485 Version
I made a local patch for me to get it to work, but a clean solution would be the definition of a new Counter Type
e.g. CGEM24ETH 

diff --git a/meters/rs485/carlogavazzi-em24.go b/meters/rs485/carlogavazzi-em24.go
index 809ab93..d089ee7 100644
--- a/meters/rs485/carlogavazzi-em24.go
+++ b/meters/rs485/carlogavazzi-em24.go
@@ -16,6 +16,10 @@ func NewCarloGavazziEM24Producer() Producer {
        /***
         * https://gavazzi.se/app/uploads/2020/11/em24_is_cp.pdf
         */
+
+       /* EM24 Ethernet has different register Mapping!
+        * https://www.gavazziautomation.com/images/PIM/OTHERSTUFF/COMPRO/EM24_E1_CP.pdf
+        */
        ops := Opcodes{
                VoltageL1: 0x00,
                VoltageL2: 0x02,
@@ -27,16 +31,30 @@ func NewCarloGavazziEM24Producer() Producer {
                PowerL2:   0x14,
                PowerL3:   0x16,
                Power:     0x28,
-               CosphiL1:  0x32,
-               CosphiL2:  0x33,
-               CosphiL3:  0x34,
-               Cosphi:    0x35,
-               Frequency: 0x37,
-               Import:    0x42,
-               ImportL1:  0x46,
-               ImportL2:  0x48,
-               ImportL3:  0x4A,
-               Export:    0x5C,
+
+               // values for ethernet
+
+               CosphiL1:  0x2e,
+               CosphiL2:  0x2f,
+               CosphiL3:  0x30,
+               Cosphi:    0x31,
+               Frequency: 0x33,
+               Import:    0x34,
+               ImportL1:  0x40,
+               ImportL2:  0x42,
+               ImportL3:  0x44,
+               Export:    0x4e,
+
+               //              CosphiL1:  0x32,
+               //              CosphiL2:  0x33,
+               //              CosphiL3:  0x34,
+               //              Cosphi:    0x35,
+               //              Frequency: 0x37,
+               //              Import:    0x42,
+               //              ImportL1:  0x46,
+               //              ImportL2:  0x48,
+               //              ImportL3:  0x4A,
+               //              Export:    0x5C,
        }
        return &CarloGavazziEM24Producer{Opcodes: ops}
 }

andig, 2023-05-09:
Es gibt bereits EM24 (`cgem24`) und EM/ET 330/340 (`cgex3x0`). Ist das jetzt nochmal eine weitere Differenzierung?

hrpv, 2023-05-09:
.fr-emoticon.fr-emoticon-img {
                                background-repeat: no-repeat !important;
                                font-size: inherit;
                                height: 1em;
                                width: 1em;
                                min-height: 20px;
                                min-width: 20px;
                                display: inline-block;
                                margin: -0.1em 0.1em 0.1em;
                                line-height: 1;
                                vertical-align: middle;
                            }
                            .fr-emoticon {
                                font-weight: normal;
                                font-family: "Apple Color Emoji","Segoe UI Emoji","NotoColorEmoji","Segoe UI Symbol","Android Emoji","EmojiSymbols";
                                display: inline;
                                line-height: 0;
                            }
                        Es sieht so aus als ob CGEX3X0 auch f&uuml;r EM24 Ethernet funktioniert, gerade getestet :-) 
die meisten Register scheinen identisch zu sein

 https://gavazzi.se/app/uploads/2020/11/em24_is_cp.pdf (https://gavazzi.se/app/uploads/2020/11/em24_is_cp.pdf" id="isPasted" rel="noopener noreferrer" target="_blank)
 https://gavazzi.se/app/uploads/2020/11/em330_em340_et330_et340_cp.pdf (https://gavazzi.se/app/uploads/2020/11/em330_em340_et330_et340_cp.pdf" id="isPasted" rel="noopener noreferrer" target="_blank)
also keine Code &Auml;nderung, nur Doku Update  mbmd_run.md

-----urspr&uuml;ngliche Nachricht-----

frosty-geek, 2023-06-07:
Scheinbar ist EM340 auch nur teilweise compatible, ich hab hier die offizielle Doku zur EM24 Ethernet variante gefunden:
https://www.enika.eu/data/files/produkty/energy%20m/CP/em24%20ethernet%20cp.pdf

hrpv, 2023-06-07:
> Scheinbar ist EM340 auch nur teilweise compatible, ich hab hier die offizielle Doku zur EM24 Ethernet variante gefunden: https://www.enika.eu/data/files/produkty/energy%20m/CP/em24%20ethernet%20cp.pdf

Mit  CGEX3X0 als Typ eingestellt funktioniert der EM24 Ethernet bei mir ohne Probleme, hast du das mal probiert?

frosty-geek, 2023-06-08:
Ja habe ich, die Basis Werte sind da, aber es fehlen diverse Werte die man sonst via Modbus TCP bekommen kann.

Der Tagesverbrauch z.b. kann aktuell nicht ordentlich bestimmt werden. Ich benutze Homeassistent mit nativen Modbus TCP Support parallel und hier fehlen die Werte einfach (siehe communication protocol aus der verlinkten Doku).

hrpv, 2023-06-08:
> Ja habe ich, die Basis Werte sind da, aber es fehlen diverse Werte die man sonst via Modbus TCP bekommen kann.
>
> Der Tagesverbrauch z.b. kann aktuell nicht ordentlich bestimmt werden. Ich benutze [...]

O.k. dann  musst du wohl selbst Hand anlegen, und den vorgeschlagenen Patch passend um weitere Register erweitern

guido4096, 2025-05-03:
Hi, I find the current solution highly confusing, especially icw evcc. Would it be ok to add indeed a new variant carlogavazzi-em24_e1  ? According to the documentation, the suffix _E1 is used identify the model. https://www.enika.eu/data/files/produkty/energy%20m/CP/em24%20ethernet%20cp.pdf

I would then also like to store the pdf's for both versions in the files section of this project:
- RS485: https://www.ccontrols.com/support/dp/CarloGavazziEM24.pdf
- Ethernet: https://www.enika.eu/data/files/produkty/energy%20m/CP/em24%20ethernet%20cp.pdf

EVCC should then also need to have explicit support for the variant EM24_E1 and with proper documentation this would all become understandable. 

If you agree, I will create a PR. I would create a complete new implementation of em24_e1 .

andig, 2025-05-03:
It might be nice to add two new names and keep the old one as alias?

guido4096, 2025-05-03:
I cna do that, although... I'm not sure it will improve with this. The EM24 seems to be the common name for the RS485 version and the EM24_E1 for the Ethernet version. I think adding EM24_E1 and leaving the EM24 as is, would be fine?

guido4096, 2025-05-03:
I created a pull request, let me know if this makes sense

PR #361 (2025-05-03): Add EM24_E1 as explicit meter to remove confusion with EM24
This is the version with adding EM24_E1

I opted to not share code with carlogavazzi-ex3x0.go ; although the code is practically the same, I think it would cause more confusion to try to share this.

Fixes: #317
closed 2025-05-03 by andig