volkszaehler/mbmd #319: show battery status

jarau-de, 2023-06-01:
Is there a way to get also battery info from Kostal Plenticore inverter?

MarkusGH, 2024-01-25:
Same question here with SolarEdge Hybrid SE10K-RWS. Installed a battery, but it seems mbmd does not query the battery related values. Querying the inverter with evcc works fine. A clarification would be great!

SchumacherFM, 2024-04-10:
I will take a look, once I have installed my battery. I have the same SE10K-RWS

ambanmba, 2025-03-01:
I just got dual battery readings working here: https://github.com/ambanmba/sunspec-monitor but I have nowhere near the skills needed to implement this in mbmd which is what I would prefer to use. Maybe someone can use the sunspec-monitor code to help with updating mbmd?

ambanmba, 2025-03-04:
This project also has working battery support: https://github.com/DerOetzi/solaredge2mqtt?tab=readme-ov-file

Someone with the skills could update mbmd

ambanmba, 2025-03-05:
This has even a better battery implementation: https://github.com/s53zo/SolarEdge_to_MQTT

ambanmba, 2025-03-11:
I've managed to get it to read from my two batteries and created a pull request here: https://github.com/volkszaehler/mbmd/pull/357

I'm running two {SolarEdge SE Energy Bank  SMCU 1.1.53 DCDC 1.1.33 BMS 1.0} batteries and it seems to be reading them fine now.

This is what's coming across on MQTT:

se-bat1-1
BatteryMaxPowerCharge = 5000.000
BatteryRatedCapacity = 9.700
BatteryTemperature = 28.952
BatteryVoltage = 407.423
BatteryMaxPowerDischarge = 5000.000
BatteryCurrent = 1.706
BatteryPower = 693.000
BatteryAvailableEnergy = 10.009
BatterySOH = 100.000
BatterySOC = 30.942
se-bat1-1-1
BatteryMaxPowerCharge = 5000.000
BatteryRatedCapacity = 9.700
BatteryTemperature = 28.642
BatteryMaxPowerDischarge = 5000.000
BatteryVoltage = 403.911
BatteryCurrent = 1.705
BatteryPower = 691.000
BatterySOH = 100.000
BatteryAvailableEnergy = 9.974
BatterySOC = 30.883