volkszaehler/mbmd #320: SUNSPEC Goodwe ModBus liesst EXPORT Wert nicht.

helix0712, 2023-06-02:
Hallo,

ein 
```mbmd inspect -a /dev/ttyUSB2  -b 9600 --comset 8N1 -d SUNS:247``` zeigt mit alle Werte an, welche der Inverter ausspuckt (siehe unten).

Mittels mbmd API bekomme ich aber nur Model103 Werte (ausgelesen mit http://192.168.19.2:9090/api/last)
```
{"SUNSPEC1.247":{"Timestamp":"2023-06-02T09:17:21.926115857+02:00","Unix":1685690241,"ApparentPower":1038.000000,"Cosphi":0.993000,"Current":1.500000,"CurrentL1":1.500000,"CurrentL2":1.500000,"CurrentL3":1.500000,"DCCurrentS1":0.900000,"DCCurrentS2":0.900000,"DCPowerS1":436.000000,"DCPowerS2":492.000000,"DCVoltageS1":484.600000,"DCVoltageS2":546.900000,"Export":3668.400000,"Frequency":50.010000,"HeatSinkTemp":40.400000,"Power":943.000000,"ReactivePower":-123.000000,"VoltageL1":230.700000,"VoltageL2":230.100000,"VoltageL3":233.600000}}
```
Muss ich noch etwas anders konfigurieren?
mbmd wird aus GIT Source kompiliert und mit mbmd.yaml gestartet:
```
# REST api, use 127.0.0.1 to restrict to localhost
api: 0.0.0.0:9090

# mqtt config
mqtt:
  broker: localhost:1883
  topic: mbmd
  user:
  password:
  clientid: mbmd
  qos: 0
  homie: homie

# influxdb_v1 config
influx:
  url: http://localhost:8086
  database: data
  measurement: mbmd
  user:
  password:

# influxdb_v2 config
#influx:
  #url: http://localhost:8086
  #database: data
  #measurement: mbmd
  #organization:
  #token:

# adapters are referenced by device
adapters:
- device: /dev/ttyUSB2
  baudrate: 9600
  comset: 8N1 # "8E1" needs be quoted as string or will error

# list of devices
devices:
- name: Goodwe
  type: sunspec
  id: 247
  subdevice: 0 # use subdevice to access SunSpec subdevices
  adapter: /dev/ttyUSB2
```

Ausgabe von inspect:
```
--------- Model 1 common ---------
Mn             GOODWE    string
Md         GW8K-DT       string
Opt                      string
Vr           V1.15.15    string
SN   XXXXXXXXXXX    string
DA             247.00    uint16
Pad            0x8000    pad
--------- Model 103 inverter ---------
A                    0.90    uint16
AphA                 0.90    uint16
AphB                 0.90    uint16
[... 463 lines cut ...]
N                       2    count
TmsPer                n/a    uint16
-- Block 1 --
ID                 1.00    uint16
IDStr                      string
DCA                0.30    uint16
DCV              528.70    uint16
DCW              158.00    uint16
DCWH                NaN    acc32
Tms                 n/a    uint32
Tmp                 n/a    int16
DCSt                n/a    enum16
DCEvt               n/a    bitfield32
-- Block 2 --
ID                 2.00    uint16
IDStr                      string
DCA                0.40    uint16
DCV              521.40    uint16
DCW              208.00    uint16
DCWH                NaN    acc32
Tms                 n/a    uint32
Tmp                 n/a    int16
DCSt                n/a    enum16
DCEvt               n/a    bitfield32
```

TRIROG, 2023-07-23:
Just figured the same thing, not all data shown on inspect is shown on run.

andig, 2026-09-30:
This is working as designed. `inspect` dumps every raw SunSpec register, while `run` only publishes the points that are mapped to measurements (models 101/103/111/113, 160, 201/203/211/213, 124). Your API output already contains all mapped values of model 103 and both MPPT strings from model 160; the remaining models (120/121/122/123/126/132) are nameplate, settings, status and curve registers, not measurements. Missing temperature/status points are tracked in #326.

🤖 Generated with [Claude Code](https://claude.com/claude-code)

closed 2026-09-30 by andig