volkszaehler/mbmd #325: Feature request: make meter configurable

DocZoid, 2023-06-26:
I spent quite some time installing a docker container and getting mbmd running just to find out that it only works with the predefined meters (at least that is my current understanding). Please point that out in the documentation.
I was actually hoping to be able to configure the addresses, lengths and meanings of my meter in the yaml config file. That way the daemon could be easily extended to future meters without the need of new releases.
I have a B+GE-TECH WS100-1943. Some addresses are in the range of a ORNO3p, but for some reason I cannot use that configuration.

chrostek, 2025-01-04:
https://github.com/volkszaehler/mbmd/pull/353
does not add your requested feature, but adds the meter you mentioned

DocZoid, 2025-01-04:
Thank you very much, I will give it a try when I have the time.

andig, 2026-09-30:
Sorry for letting this sit for so long without a resolution. The request is valid, but nobody has picked it up in the meantime and there are no plans to implement it from the maintainer side, so I am closing this to keep the tracker honest. If it is still relevant, a PR is very welcome and we will happily reopen.

🤖 Generated with [Claude Code](https://claude.com/claude-code)

closed 2026-09-30 by andig