volkszaehler/mbmd #327: Inepro PRO380 MOD old version

daemonenstall, 2023-08-13:
Hello @andig 

I have an old version of the PRO380 MOD from Inepro and it provides the same features but the Modbus registers are different. Right now I don't use mbmd, but in the near future I would like to use EVCC. At the moment the meter is connected to my home assistant.

The old manual can be found here: https://vallin.lv/wp-content/uploads/2017/10/PRO380-user-manual-V1-03.pdf
In the manual of the German distributor, old and new registers are mentioned side by side: https://www.kdk-dornscheidt.de/wp-content/uploads/2017/11/Bedienungsanleitung_KDK_PRO380_4205100Pro_420506Pro_S_Mbus_Modbus_V3_00.pdf

For your convenience, I have modified the registers of https://github.com/volkszaehler/mbmd/blob/master/meters/rs485/inepro.go

Since I don't know if this old version was also available as 1 phase meter, I have removed the opcodes that are only available for the 1 phase version. 

If you need someone for testing, let me know.

Best, Tom

[inepro_old_registers.go.txt](https://github.com/volkszaehler/mbmd/files/12329496/inepro_old_registers.go.txt)

andig, 2023-08-13:
Where in the pdf can I find the comparison?

daemonenstall, 2023-08-13:
The table starting at page 37. The first column represents the old registers and the second column are the registers that are already used in inepro.go
https://www.kdk-dornscheidt.de/wp-content/uploads/2017/11/Bedienungsanleitung_KDK_PRO380_4205100Pro_420506Pro_S_Mbus_Modbus_V3_00.pdf

andig, 2023-08-13:
Ah, I see. Any idea what makes the "old" version? Different make or name?

daemonenstall, 2023-08-13:
Well, that's a good question. My meter has the following versions (all can be accessed by modbus):
Hardware version: 1.04
Software version: 1.18 
Protocol version: 3.20
Since the protocol version is the same in both manuals that I have mentioned, I think the software version should be the right one.

andig, 2026-09-30:
Sorry for letting this sit for so long without a resolution. Nobody has picked up the implementation in the meantime and there are no plans to work on it from the maintainer side, so I am closing this to keep the tracker honest. If the device is still relevant to you, a PR adding it (the existing meters/rs485/*.go files serve as templates) is very welcome and we will happily reopen.

🤖 Generated with [Claude Code](https://claude.com/claude-code)

closed 2026-09-30 by andig