volkszaehler/mbmd #329: solar edge meter

frezeen, 2023-09-24:
how can i read meter SEMTR     SolarEdge SE-MTR-3Y ??

i use this command:
./mbmd run -d solaredge:1@192.168.2.59:1502 
and i have data from inverter, how can i read meter too? 
i try everything but no solution

andig, 2023-09-24:
s/:1/:2/

frezeen, 2023-09-24:
sorry but , what mean s/:1/:2/ ?

TRIROG, 2024-03-03:
> s/:1/:2/

Can you provide some additional info for better understanding?

SchumacherFM, 2024-03-24:
@TRIROG  Just change the ID from 1 to 2 or from 2 to 1. Or try 3. If the ID isn't configured you get a timeout.
The meter is usually a subdevice of the inverter.

TRIROG, 2024-03-24:
Was taking about syntax … just for better understanding

sicheste, 2024-03-26:
@TRIROG The `s/:1/:2` means "replacing `:1` with `:2`". This is for example used in the vi / vim text editor. The `s` is for _substitute_ or _search and replace_, the `/` is just a delimiter (but could be any other alphanumeric symbol) and the `:1` is the search pattern which will be replaced with the `:2`.

andig, 2026-09-30:
Reading the SolarEdge meter works by addressing it as a SunSpec subdevice, as you found: `mbmd run -a 192.168.2.59:1502 -d solaredge:1 -d solaredge:1.1` (the `.1` selects the first subdevice, i.e. the meter). Closing as answered.

🤖 Generated with [Claude Code](https://claude.com/claude-code)

closed 2026-09-30 by andig