volkszaehler/mbmd #330: Add-Request: Implementation of meter YG889 from Shanghai Yige Electric Co., Ltd. possible?

Flat-Mac, 2023-10-20:
Hello developers,
I am using a smartmeter YG889-9SY from Shanghai Yige Electric Co., Ltd. to count my solar power. Unfortunately it isn't mentioned in the list of existing meters. The register of the meter is here attached. [Register von YG889.pdf](https://github.com/volkszaehler/mbmd/files/13056740/Register.von.YG889.pdf)

It would be fine, if sometime in the future this meter would be supported.

Kind regards 

Thomas

andig, 2023-10-20:
Happy to take a PR!

Flat-Mac, 2023-10-21:
Hello Experts,
I forgot to list a picture of the registers made by "Radzio! Modbus Master Simulator".
[Register.von.YG889.pdf](https://github.com/volkszaehler/mbmd/files/13061954/Register.von.YG889.pdf)
Compared to the PDF-file above, here you can see a difference in the numbering of the register. This is the real numbering because the data make sense (compared to the data on the meter) and with the same numbering, also the graphs of the actual power shown in "ModbusScope" on a PC, are correct.

Kind regards

Thomas

Flat-Mac, 2023-10-23:
Sorry, wrong attachment!
[image]
Thomas

andig, 2026-09-30:
Sorry for letting this sit for so long without a resolution. Nobody has picked up the implementation in the meantime and there are no plans to work on it from the maintainer side, so I am closing this to keep the tracker honest. If the device is still relevant to you, a PR adding it (the existing meters/rs485/*.go files serve as templates) is very welcome and we will happily reopen.

🤖 Generated with [Claude Code](https://claude.com/claude-code)

closed 2026-09-30 by andig