volkszaehler/mbmd #334: Using mbmd with reverse-proxy

twam, 2024-02-23:
Hey!

I want to use mbmd with a reverse-proxy that maps the mbmd website to `https://hostname/mbmd`. The website however still tries to reach `wss://hostname/ws` which isn't working as the correct URL would be `wss://hostname/mbmd/ws`.

Would it be possible to adapt the URL for such use-cases?

Best regards
  Tobias

Jennes, 2024-03-20:
With this commit included, the web frontend doesn't work anymore when accessed directly with a web browser. The current chrome version then tries to access `ws://localhost:80//ws` (mind the double '/' in the path section of the URL)

My VueJS and JS knowledge is rather limited, so I wouldn't know how to insert `loc.pathname` only when it's not empty or not equal to "/" in an elegant way.

andig, 2024-03-20:
@twam could I kindly ask you for a fix?

twam, 2024-03-20:
It worked for me, but I’ll have a look.

Jennes, 2024-03-20:
@twam If it actually runs behind a proxy like nginx, everything is fine, as far as I could tell. I don't know if `loc.path` is a string or something different (as I said, I don't know much about how JS works within a browser) something like ` … + ((loc.path && loc.path != '/') ? loc.path : "") + …` could already be enough. I'm not sure if the outer parenthesis are necessary or not, maybe the ternary operator binds stronger than `+`

twam, 2024-03-20:
With 'worked for me', I meant that I tested also a setup without proxy, i.e. directly called the web server provided by mbmd. Just rechecked and worked fine using Safari :)

Nevertheless, created a PR with removes any trailing slashes from `loc.pathname` which hopefully should fix the issue. At least works for me with/and without proxy with Safari and Chrome.

PR #335 (2024-02-24): fix: fix path to websocket when running behind reverse-proxy
Fixes #334.