volkszaehler/mbmd #339: disable homie via yaml-config

astrakid, 2024-03-06:
hi,
i am trying to disable homie-export via mqtt.
in yaml-file i tried 
[...]
homie:
[...]
but that doesn't stop data being sent with topic "homie". i tried to comment out that line as well without success.
how would i disable exporting data to homie-topic?

regards,
andre

Caibuk, 2024-03-06:
That's something I tried as well with no succes. I don't need the homie topic in my eco system.

SchumacherFM, 2024-03-24:
Set any occurence of `mqtt.homie` in yaml or `--mqtt-homie` to an empty string and it will be disabled.
for the cli it looks like `--mqtt-homie=`

andig, 2024-03-24:
Happy to take a PR that makes Homie opt-in or even removes it.

Caibuk, 2024-05-03:
Thanks to @SchumacherFM's hint I got finally rid of the homie topic defining this in my mbmd.yaml:
```
mqtt:
  homie: ''
```

closed 2026-09-30 by andig