volkszaehler/mbmd #340: data from SDM120M incomplete

sicheste, 2024-03-10:
If I add the SDM120M as `SDM220` to the config, I will not get all values. Especially the `Power (W)` is missing. If I add it as a `SDM` I get this value.

As `SDM220`:
[image]

As `SDM`:
[image]

Btw: What is the `name` field in the config for? The name (here `SDM2201.3` and `SDM1.3` is constructed by type + position in the alphabetically sorted list of sensors + `.` or `-` + bus id. Thus the names in the API and MQTT are unstable.

sicheste, 2024-03-11:
I just recognized that the current docker image is already 4 years old. I will try in the next days to build my own one using the current master branch and give it a try. Maybe it is time for a new version :)

sicheste, 2024-03-14:
So, I had some time. I have created a new docker image locally, based on the current master branch. Running `mbmd scan` was not detecting the SDM120M. But manually giving it as SDM220:3 (it is number 3 on the bus) it works. Data in MQTT and on the web interface are fine.

So, still open:
* The detection is not working (at least here).
* The names are constructed using IDs, which are based on the alphabetical order of the meters. This makes the naming unstable when adding / removing meters.

isarrider, 2024-06-20:
https://hub.docker.com/r/isarrider/mbmd/tags
try mine...

isarrider, 2024-06-20:
nah, I just wanted to give @sicheste and others a docker image that isnt old...
(actually thats why I started also building mine, bc there is no newer)

andig, 2024-06-20:
Thank you. Would be happy for build workflow PR- don't have the time...

isarrider, 2024-06-20:
I know the feeling but I am always hoping for the young IT Students or similar...

andig, 2026-09-30:
The missing Power value came from the outdated Docker image; SDM220 reads Power since #217 and SDM120 has its own `SDM120` type since #377. Current images are now published via #399. `mbmd scan` over TCP only probes SunSpec by design (see docs/mbmd_scan.md). Using the configured `name` as device key is tracked in #369, so closing this one.

🤖 Generated with [Claude Code](https://claude.com/claude-code)

closed 2026-09-30 by andig