volkszaehler/mbmd #346: No reconnect to MQTT after restart MQTT-Server

Fowow, 2024-05-29:
Hello,

after rebooting the remote MQTT, the MQTT data on the broker disapears.  I need to restart  MBMD and then , it connects successful and sends data.

I use a Node-Red MQTT-Broker.

andig, 2024-05-29:
Sounds like reconnect handler needs be added.

Fowow, 2024-05-30:
OK, how can this be done? This is important for me because I sometimes forget to manualy restart the MBMD Daemon for reconnecting :-(

andig, 2024-05-30:
Seems reconnect is already enabled. No idea why this wouldn't work. It should.

Fowow, 2024-05-31:
I use the "aedes broker" included in NODE-RED. I have connected many clients to access this broker.
If i deploy some changes in Node-Red all devices re-connets, except the MBMD. I need to go t my raspi and restart the MBMD-Service.

Looks that there is a BUG in the "reconnect handler"?

Fowow, 2024-07-11:
Some days ago, I restarted my NodeRed-MQTT Server and I forgot the MDMD daemon to restart.
All 10 MQTT clients have successful re-connect exept the MDMD-Daemon. :-(

It is possible that this bug will be fixed because I have los important accounting data from my solar-devices.

andig, 2024-07-11:
> Seems reconnect is already enabled. No idea why this wouldn't work. It should.

This will need someone to debug why the reconnect isn't working.

Fowow, 2024-07-12:
This would be great, I realy like MBMD :-)
It is easy for you to debug, just install Node-Red and "aedes broker". After connecting everything you can restart the Node and you see the disconnect.

everte, 2025-09-18:
I am experiencing the same issue with mosquitto broker. It would be amazing if the automatic reconnect could work, I often have gaps in my measurements due to small network hiccups or restarting of servers/services and needing to do manual intervention.

andig, 2025-09-18:
https://github.com/volkszaehler/mbmd/issues/346#issuecomment-2223405560