volkszaehler/mbmd #347: Permission denied for /dev/ttyUSB0 when use Docker

chrissooo, 2024-09-01:
Hi,

My Setup:
- Raspberry Pi 4 with fresh Bookworm installation
- USB modbus Dongle on /dev/ttyUSB0
- ORNO 515

Start docker with following comman (also try privillage):
`docker run --privileged -p 8081:8080 --device=/dev/ttyUSB0 volkszaehler/mbmd run -a /dev/ttyUSB0 -d ORNO1p:1 --comset 8E1`

After start following errors i see:
`2024/09/01 10:20:49 mbmd 0.13 (fa6b574)
2024/09/01 10:20:49 config: creating RTU connection for /dev/ttyUSB0 (9600baud, 8E1)
2024/09/01 10:20:49 config: declared device ORNO1p:1.0
2024/09/01 10:20:49 httpd: starting api at 0.0.0.0:8080
2024/09/01 10:20:49 initialized device ORNO1p1.1: {ORNO1p ORNO1p ORNO WE-514 & WE-515    0}
2024/09/01 10:20:49 device ORNO1p1.1 did not respond (1/3): read failed: permission denied
2024/09/01 10:20:50 device ORNO1p1.1 did not respond (2/3): read failed: permission denied
2024/09/01 10:20:51 device ORNO1p1.1 did not respond (3/3): read failed: permission denied
2024/09/01 10:20:52 device ORNO1p1.1 is offline
2024/09/01 10:20:54 device ORNO1p1.1 is offline - reactivating
2024/09/01 10:20:54 device ORNO1p1.1 did not respond (1/3): read failed: permission denied
2024/09/01 10:20:55 device ORNO1p1.1 did not respond (2/3): read failed: permission denied`

When i download the nbmd binary fpr my pi and start it with the same Parameter for example:
`./mbmd run -a /dev/ttyUSB0 -d ORNO1p:1 --comset 8E1`
its work without problems.

can you help me, i make something wrong maybe?

chrissooo, 2024-09-01:
Al little Update. i have build the Docker Image new from Dockerfile from this repository, and then its work. maybe becource newer version of alpine? Is it possible to rebuild it to get a working Version from docker pull?

closed 2025-02-08 by andig

Caibuk, 2025-02-08:
Hi @andig ,
That’s great to hear! I’d love to use Docker instead of my own custom-built binaries.

As far as I understand, the Docker publish automation is triggered by a new tag. When will a new tag be added or a new version released? Or am I missing something?

Thank you much for still contributing to this repo!