volkszaehler/mbmd #364: Support for Goodwe ET hybrid inverters

air-ii, 2025-05-22:
Hi, I'd like to add support for Goodwe hybrid inverters that aren't Sunspec compliant. This is more a question on how to approach it best in terms of structure. I was thinking of creating a tcp package within meters and implement (also extend) supported list of measurements.

I would base the implementation on the register list found here: 
https://github.com/marcelblijleven/goodwe/blob/master/goodwe/et.py

Amy suggestions?

andig, 2025-05-22:
> I was thinking of creating a tcp package

Does the inverter speak modbus? If the inverter is not modbus compliant then I'd refrain from adding it. I don't think we want to start building any amount of specific integrations in mbmd.

air-ii, 2025-05-22:
Yes, it does speak modbus.

air-ii, 2025-05-23:
I'm closing this issue. As a quick solution for myself, I've added an rs485 device that serves its purpose. I won't be creating a PR for this though, as this quick hack doesn't seem right from the tool's design standpoint.

As a food for thought, this particular inverter is an all in one solution with an integrated grid energy meter, which makes it difficult to expose all the readings (PV + grid) under one device. In my case I am reading only PV + battery data, as I'm monitoring the grid using an Inepro meter anyway.

Btw, I was not able to make enumer work with added measurements (for battery), so I ended up altering the generated file semi manually.

closed 2025-05-23 by air-ii